justcallmekoko/ESP32Marauder #742: Update WifiScan:RunSetup() watch_models list to include all apple watches

1cyb3rpunk, 2025-05-02:
**Is your feature request related to a problem? Please describe.**
This is a feature request expand the type of apple watches detected

**Describe the solution you'd like**
Looking to add to the current array

**Describe alternatives you've considered**
None

**Additional context**
Adding new watches based on apple documentation. identifiers. 

watch_models = new WatchModel[41] {
    {0x1A, "Fallback Watch"},
    {0x01, "White Watch4 Classic 44m"},
    {0x02, "Black Watch4 Classic 40m"},
    {0x03, "White Watch4 Classic 40m"},
    {0x04, "Black Watch4 44mm"},
    {0x05, "Silver Watch4 44mm"},
    {0x06, "Green Watch4 44mm"},
    {0x07, "Black Watch4 40mm"},
    {0x08, "White Watch4 40mm"},
    {0x09, "Gold Watch4 40mm"},
    {0x0A, "French Watch4"},
    {0x0B, "French Watch4 Classic"},
    {0x0C, "Fox Watch5 44mm"},
    {0x11, "Black Watch5 44mm"},
    {0x12, "Sapphire Watch5 44mm"},
    {0x13, "Purpleish Watch5 40mm"},
    {0x14, "Gold Watch5 40mm"},
    {0x15, "Black Watch5 Pro 45mm"},
    {0x16, "Gray Watch5 Pro 45mm"},
    {0x17, "White Watch5 44mm"},
    {0x18, "White & Black Watch5"},
    {0x1B, "Black Watch6 Pink 40mm"},
    {0x1C, "Gold Watch6 Gold 40mm"},
    {0x1D, "Silver Watch6 Cyan 44mm"},
    {0x1E, "Black Watch6 Classic 43m"},
    {0x20, "Green Watch6 Classic 43m"},
    {0x21, "Apple Watch (1st Gen)"},
    {0x22, "Apple Watch Series 1"},
    {0x23, "Apple Watch Series 2"},
    {0x24, "Apple Watch Series 3"},
    {0x25, "Apple Watch Series 4"},
    {0x26, "Apple Watch Series 5"},
    {0x27, "Apple Watch Series 6"},
    {0x28, "Apple Watch Series 7"},
    {0x29, "Apple Watch Series 8"},
    {0x2A, "Apple Watch Series 9"},
    {0x2B, "Apple Watch Series 10"},
    {0x2C, "Apple Watch SE (1st Gen)"},
    {0x2D, "Apple Watch SE (2nd Gen)"},
    {0x2E, "Apple Watch Ultra"},
    {0x2F, "Apple Watch Ultra 2"}
};

justcallmekoko, 2025-05-09:
If you are interested, You can submit this as a PR so you can be included as one of the contributors. I would be more than happy to see your name show up on the list.

If you do submit a PR, go off of the develop branch

1cyb3rpunk, 2025-05-10:
Thanks bro! I have some other functions I would love to contribute as well.

Sent from [Proton Mail](https://proton.me/mail/home) for iOS

closed 2025-05-14 by 1cyb3rpunk

1cyb3rpunk, 2025-05-16:
Hey bro, can you add me as a collaborator so I can submit pr's? I tried to submit one and it told me I need to be added as a collaborator.

TIA

Sent with [Proton Mail](https://proton.me/mail/home) secure email.