justcallmekoko/ESP32Marauder #743: The TFT screen is landscape instead of portrait

Yesnowway, 2025-05-03:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Tazo-ai, 2025-05-04:
Which display are you running? st7789 or ili9341?

coccobello27, 2025-05-04:
i have esp32-2432s028 module, touch is ok but image is rotate 90` - 

[image]

khaleeqmalik786786, 2025-05-09:
this is st7789

coccobello27, 2025-05-10:
2.8-inch color TFT display screen with ILI9341 driver chip
Display resolution: 240x320px with resistive touchscreen

[ESP32-2432S028R](https://randomnerdtutorials.com/cheap-yellow-display-esp32-2432s028r/)

ToxxikHalo, 2025-05-10:
Fwiw I have the same issue on same device...

stevedee78, 2025-05-16:
I (CYD2USB) have the same problem!

x0SiN0x, 2025-05-29:
came here to report the same thing, quick bump to this having the same issue.  CYD 2432S028R (CYD2USB)

khaleeek, 2025-05-30:
So this issue is not with any thing this issue with st7789 the firmware support ili9341 but your display is st7789 so you can use st7789 firmware

justcallmekoko, 2025-05-30:
https://github.com/justcallmekoko/ESP32Marauder/pull/776

closed 2025-05-30 by justcallmekoko

x0SiN0x, 2025-05-30:
FWIW its working here on mine, flashed shortly after release 1.6.1