justcallmekoko/ESP32Marauder #753: horizontal screen

Idezcs, 2025-05-12:
Hey, I have a problem. The screen is horizontal instead of vertical. I don't know what to do. I tried with PlatformIO.  I have tried new and old software but it still does not working.
I have:
 esp 32 wroom 32u
Screen st7789
 [image]

justcallmekoko, 2025-05-12:
The firmware is not currently written to accommodate this screen. You will need to configure your own instance of the TFT_eSPI library

closed 2025-05-12 by justcallmekoko

ToxxikHalo, 2025-05-12:
Hey koko, apologies if you addressed this already.. is it the same answer for the CYDs that have this same issue??? Thanks

Idezcs, 2025-05-13:
then can I still use platformIO? or do I have to use arduino IDE?

Idezcs, 2025-05-13:
then can I still use platformIO? or do I have to use arduino IDE? > The firmware is not currently written to accommodate this screen. You will need to configure your own instance of the TFT_eSPI library