justcallmekoko/ESP32Marauder #762: Request for a custom firmwae

theminer800, 2025-05-19:
Hi there I want a custom firmware for my ESP32-3248S035R 3,5” display I have been working on this project for 3 months with no success I would like the firmware to have these

WiFi Tools:
• WiFi Deauth Attack
• Beacon Spam
• Evil Portal
• WiFi Scanner
• WiFi Access Point Mode
• WiFi Connection Mode
• Packet Monitor
• Raw Packet Sniffer
• Scan Host
• TCP Listener
• TCP Client
• Telnet Server
• SSH Server
• DHCP Snooping
• Web Listener
• WireGuard VPN Configuration
• MAC Address Randomizer (WiFi)

Bluetooth (BLE) Tools:
• BLE Scanner
• BLE Advertiser
• BLE Spoofer (Samsung)
• BLE Spoofer (iOS)
• BLE Spoofer (Android)
• BLE Spoofer (Windows)
• BLE Spoofer (All Devices)
• BLE Media Commands
• Bad BLE Attacks
• Bluetooth Serial Terminal
• Bluetooth MAC Address Randomizer

RF & Sub-GHz Tools:
• RF Spectrum Analyzer
• Square Wave RF Generator
• Jammer Full (All RF)
• Jammer ITMT (Targeted RF)
• Sub-GHz Frequency Tuning
• Sub-GHz Scan & Copy
• Sub-GHz Custom Signal Sender

IR (Infrared) Tools:
• IR Signal Learner
• IR Signal Spammer
• TV Beacon Mode
• Custom IR Command Storage
• IR Hex Viewer

⸻

FM Tools:
• FM Radio Controller
• FM Frequency Scanner
• FM Spectrum Analyzer

⸻

GPS Tools:
• GPS Tracker Interface
• GPS Logger
• GPS Viewer
• Bluetooth GPS Bridge (receive GPS via phone over BLE)

⸻

RF24 Tools (Your Additions):
• NRF24L01 Module Support
• NRF24 Data Receiver
• NRF24 Data Display on Screen
• NRF24 Serial Bridge Option

⸻

System Tools:
• File Browser (SD Card)
• QR Code Generator
• Web UI (Full HTML Control Panel)
• JS Interpreter (Run Custom JavaScript)
• OpenHaystack Tracker
• Megalodon Detector
• BadUSB Emulator
• iButton Emulator
• Timer Utility
• Clock Display
• Logs Viewer
• Brightness Control
• Config Menu
• Startup Logo/Name Editor
• System Reboot
• OTA Update Server
• Web Flasher Compatibility
• Persistent Settings Save
• Serial Command Console

⸻

Display & UI (for ESP32-3248S035R):
• Full GUI Menu System (Touch-Based)
• Touchscreen Compatibility (3.5” ILI9488)
• UI Scaling for 480x320 Resolution
• Custom Menu for GPS Viewer
• Custom Menu for NRF24 Live Data
• IR Control Menu
• BLE Command Menu
• Configurable Button Colors and Layout
• Real-Time Data Rendering

⸻

Custom Additions :
• NRF24L01 Data Handling (live telemetry or messages)
• Bluetooth GPS Input from Phone
• Full integration into main menu
• Compatibility with ESP32-3248S035R with no external GPIO header
• Everything bundled into one PlatformIO project
• Output also provided as ready-to-flash .bin

justcallmekoko, 2025-05-19:
Sounds like you need to go to fiverr

closed 2025-05-19 by justcallmekoko

theminer800, 2025-05-19:
Give me a link to fiverr

theminer800, 2025-05-19:
> fiverr

Give me a link to fiverr