justcallmekoko/ESP32Marauder #766: .html EP files not showing up after v6 1.5.1 firmware update

philliproach, 2025-05-21:
I recently got my first Marauder v6 and was still on a version form before the new update. I installed some .html files to try out the EP and it worked perfectly. I decided to update the firmware to 1.5.1 and now no files show up when I go to start Evil Portal.

justcallmekoko, 2025-05-21:
Can you provide some images of the root of your SD card as well as the display when you go to select an HTML file and also start evil portal? I am unable to reproduce this error on my v6 running v1.5.1

philliproach, 2025-05-21:
Thanks for the reply. The sd card icon is red so it’s like it isn’t detecting that there’s an sd card. When I updated the firmware I did an sd card update and it worked fine before, but after I updated it now it’s not detecting an sd card. I can get pics later today

justcallmekoko, 2025-05-21:
Sounds like you used the wrong bin file to update your hardware and now you are going to need to use either the FZEE flasher or the spacehuhn web updater. If you use the wrong v6.X bin file, the SD card pin configuration is different which would cause the exact issue you are experiencing.

philliproach, 2025-05-21:
I used esp32_marauder_v1_5_1_20250519_v6.bin for the firmware update. How could I use FZEE flasher to update the firmware? I’ve messed with FZEE flasher before and there’s options for update -s updates from sd card which in this case won’t work, and there’s update -w which doesn’t do anything.

closed 2025-05-21 by justcallmekoko

philliproach, 2025-05-21:
I appreciate you helping resolve my problem with the sd card not working, but I’m not sure how to use FZEE flasher to update the firmware. What command do I enter to make it update?

justcallmekoko, 2025-05-21:
Please use the [Update Firmware](https://github.com/justcallmekoko/ESP32Marauder/wiki/update-firmware) documentation. It has the link to the FZEE flasher as well as documentation for other means of updating your firmware