justcallmekoko/ESP32Marauder #773: Screen freeze while SD card inserted

solarecho10, 2025-05-28:
I’m experiencing an issue which started when I upgraded from V1.4.5 to 1.5.0.

My screen become unresponsive while I have an SD card inserted. The screen will become responsive immediately after removing the SD card.

Troubleshooting I have conducted:
1. Erased and reflashed the Marauder
2. Erased and downgraded from v1.5.0 to v1.4.5
3. Erased and upgraded from 1.4.5 to 1.5.1
4. Ensured I was using a 32gb micro SD (before this issue I could use a 200gb micro SD)
5. Ensured the SD card was formatted to FAT32
6. Connected through CLI, I’m able to operate the marauder perfectly however nothing will save to the SD card.
6.1 Upon initial boot the SD card will mount and initialise, the marauder will attempt to open files within the SD card but this won’t work.
6.2 The marauder has the ability to create but not read files
7. Inspected the hardware - no signs of issues

Potential issue:
From my low knowledge in this technology I believe this to be hardware. Some quick googling suggested it could be an error with shared pins. But it’s unique to my marauder from the looks of it.

Marauder v6 with awok GPS module.

Puchacz46, 2025-05-29:
I have the same problem and but I have this hardware on 2 esp 32

pablosguajardo, 2026-05-10:
In my case, I solved it with the comment in this thread: https://github.com/justcallmekoko/ESP32Marauder/issues/1104

closed 2026-10-18 by github-actions[bot]