justcallmekoko/ESP32Marauder #783: EAPOL and packet monitor still causing a crash in 1.6.1

Matizek1, 2025-05-31:
Hi there,the crashing after using the packet monitor or EAPOL/PMKID scan still isn't fixed.The 2.8 inch CYD still crashes after using the options a while.
Steps to reproduce:
1.After booting up the CYD go to Wifi then sniffers and choose EAPOL/PMKID scan or Packet monitor option
2.Use them a while,Deauth with other device so that the marauder can catch the deauths
3.After using it just leave the option and it will show up the Wifi category without the options for like 1 second.And there's the crash.
As i said earlier the issue is on the latest version.

Hope you fix it.
Regards!

justcallmekoko, 2025-05-31:
I am unable to reproduce this error. Can you provide a video of this happening with your process from start to finish and also show a picture of your settings.

Matizek1, 2025-06-01:
> I am unable to reproduce this error. Can you provide a video of this happening with your process from start to finish and also show a picture of your settings.

[image]

Matizek1, 2025-06-01:
> I am unable to reproduce this error. Can you provide a video of this happening with your process from start to finish and also show a picture of your settings.

https://github.com/user-attachments/assets/e2d211b6-85f4-4801-90cf-fff973801031

I used the option for like 2 minutes before i started recording this.

justcallmekoko, 2025-06-02:
What happens if you don't wait 3 minutes. What if you wait a couple of seconds?

closed 2026-10-18 by github-actions[bot]