justcallmekoko/ESP32Marauder #806: does not support esp32-s3 N16R8?

2008-joaopedro, 2025-06-15:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Marauder (please complete the following information if applicable):**
 - Firmware version: [e.g. v0.9.17]
 - Hardware version: [e.g. OG] 

**Desktop (please complete the following information if applicable):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information if applicable):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

e1ud, 2025-08-30:
Hello. Im also struggling to flash Marauder on mz ESP32/S3 N16R8. Tried to flash it with Arduino IDE, esptool and https://fzeeflasher.github.io/ but no luck. 

Tried all different things like dio, cio, 40, 80Mhz, tried different boot_app0.bin, bootloader.bin, partitions.bin but no luck.
On website I tried to flash ESP32-S3 Multiboard, also tried ESP32-WROOM but no luck. Still getting errors and loop loading. Cant figure it out.

Last thing I tried is to erase and install ESP32-S3 Multiboard from website. All went successfully but Screen Monitor showing this:
`
E (329) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037769c
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x43c
load:0x403b6000,len:0xbd0
load:0x403ba000,len:0x29c8
entry 0x403b61d8
E (240) esp_core_dump_flash: No core dump partition found!
E (240) esp_core_dump_flash: No core dump partition found!
E (6) psram: PSRAM ID read error: 0x00ffffff,Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x420c28d0: effe8188 b80008e0 ad6b8c92
 Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (IDLE) 
Core  0 register dump:
PC      : 0x40056f60  PS      : 0x00060036  A0      : 0x82052ffa  A1      : 0x3fcf76f0  
A2      : 0x3fcf7700  A3      : 0x3c0ec710  A4      : 0x00000060  A5      : 0x3fcf7720  
A6      : 0x3c0ec65f  A7      : 0x3c0ec668  A8      : 0x00000000  A9      : 0x00000001  
A10     : 0x60000000  A11     : 0x00000001  A12     : 0x420c28d6  A13     : 0x3fcf78a0  
A14     : 0x00000000  A15     : 0x3fc9b948  SAR     : 0x0000000a  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000003  

Backtrace: 0x40056f5d:0x3fcf76f0 |<-CORRUPTED

`

There was different things going on but this is the lat one Im getting. 
 
Any tips and tricks to reslove it and install Marauder on my ESP32-S3-N16R8?

Aryann019x, 2026-09-25:
Have esp32s3 2.8 TFT ILI9341/ST789 .bin files which i pushed on my web flasher, try the prebuilt bins on web flasher
Here it's live https://esp-web-flasher.runs-on.dev/