justcallmekoko/ESP32Marauder #819: How to get to the CLI on a CYD with 2 USBs

drai8192, 2025-07-02:
CYD 2.8 inch with both type C and Micro USB
I tried both with Launcher and burning only the Marauder firmware, same result.
Firmware version: 1.7.1
I'm trying to connect to the PC, but communication gets blocked after the line about battery monitor, nothing I can send or do, just disconnect.
This is the output, I just edited my board MACs:

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1100
ho 0 tail 12 room 4
load:0x40078000,len:12308
load:0x40080400,len:3076
entry 0x400805ec
E (570) esp_core_dump_flash: N<0x7f><0x06><0x1a><0xbd><0xc9><0x95><0x81>dump partition found!
E (570) esp_core_dump_flash: No core dump partition found!
E (34) psram: PSRAM ID read error: 0xffffffff
ESP-IDF version is: v4.4.5
Shutting down BLE
Successfully set AP MAC: EDITED
Successfully set STA MAC: EDITED
Evil Portal Found 0 HTML files
Checking for battery monitors...

Also, I can't seem to get the CYD to run headless with the BOOT button, is there any other way?

drai8192, 2025-07-02:
Seeing the log from another issue, I thought of trying with the GPS attached and it looks like it's getting past the battery monitor... Now it scrolls lines of <break x 3_digit_number>

$GPRMC,114843.00,V,,,,,,,020725,,,N*74

$GPVTG,,,,,,,,,N*30

$GPets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1100
ho 0 tail 12 room 4
load:0x40078000,len:12308
load:0x40080400,len:3076
entry 0x400805ec
E (570) esp_core_dump_flash: N<0xfe><0x9e><break>
<0x1a><0xbd><0xc9><0x95><0x81>dump partition found!
E (570) esp_core_dump_flash: No core dump partition found!
E (34) psram: PSRAM ID read error: 0xffffffff
ESP-IDF version is: v4.4.5
Shutting down BLE
Successfully set AP MAC: EDITED
Successfully set STA MAC: EDITED
Evil Portal Found 0 HTML files
Checking for battery monitors...
<break x 032>
<break x 032>
<break x 032>
<break x 032>
<break x 032>
<break x 032>

nicolinux, 2025-08-27:
Same problem here. I would love to test the pkimd capture and the CYD has a sd card slot.

closed 2026-10-18 by github-actions[bot]