justcallmekoko/ESP32Marauder #830: touch screen out of alignment.

troy967, 2025-07-10:
**Describe the bug**
The menu is out of alignment relative to where you touch the screen. I have a CYD 2432S028 2USB marauder and am using the 1.7.2 firmware. touch is off center.

**To Reproduce**
Steps to reproduce the behavior:
Try tap directly on menu item. I found it was more like a joypad.

**Expected behavior**
A clear and concise description of what you expected to happen.
Tap right on menu it to work properly.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Marauder (please complete the following information if applicable):**
 - Firmware version: 1.7.2
 - Hardware version: CYD 2342S028 2USB

**Desktop (please complete the following information if applicable):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information if applicable):**
na

**Additional context**
Add any other context about the problem here.

chemary, 2025-07-11:
Are you sure is out of alignment? I noticed after thinking the same that there are 3 regions (you can see marks on the edges) touching upper region is key-up, middle key-press and down key-down.

madtv, 2025-07-13:
I have the exact same issue. If the solution is the 3 regions - this is absolutely confusing.

Jymm, 2025-07-16:
Same here.  Touch screen used to work intuitively (i.e. touch the menu item with a stylus, etc. to select it).  Upgraded firmware to 1.7.2 and now it acts weirdly.  Part of the screen is 'up', part is 'down' and somewhere is 'select' but it's not consistent, especially for menus that fill the whole screen.  Tried retro'ing back to 1.6.0 and got the same behavior.  Any easy way to enable to touch screen to act like a tap-to-select touchscreen?

justcallmekoko, 2025-07-16:
Yea this isn't an issue. This was a feature request from the community. There are now up (top), down (bottom), and select (center) touch regions on the display marked by gray brackets that you now use to navigate the menu

closed 2025-07-16 by justcallmekoko

troy967, 2025-07-28:
Now that's been made clear, I get it.  Thanks to those involved.  JT :)