justcallmekoko/ESP32Marauder #855: GPX File Time Data Syntax Correction

H4W9, 2025-08-01:
**Describe the bug**
There is a minor flaw with the Time Data syntax in the GPX files generated by the GPS Tracker in Marauder. Some GPX viewers don't accept single-digit numbers in the time data, so storing single-digit numbers like 1 as double-digit... 01 would resolve the issue. Could you please fix?

Example:
Current syntax:                    <time>2025-7-30T15:12:4Z</time>
Fixed syntax:                        <time>2025-07-30T15:12:04Z</time>
Single digits in the Time Data should be stored with a zero in front of them, turning them into a double-digit number.

**To Reproduce**
Any GPS Track file generated from Marauder v1.8.3 will only work with some GPX viewers

**Expected behavior**
GPS Track file generated from Marauder v1.8.3 will work with all or most GPX viewers

**Marauder (please complete the following information if applicable):**
 - Firmware version: v1.8.3
 - Hardware version: 6.1 (Awok Dual Touch V3)

**Additional context**
GPXsee is one of the GPX file viewers I used to test the GPX files saved by Marauder.

justcallmekoko, 2025-08-04:
Fixed in https://github.com/justcallmekoko/ESP32Marauder/commit/b490b5968ffc26a1f942871ec62553928648199e.

Will be implemented in the next release

closed 2025-08-04 by justcallmekoko

H4W9, 2025-08-04:
Glad to Hear! Thanks for adding a fix. Keep up the hard work on this impressive project.