justcallmekoko/ESP32Marauder #918: 2.4 ILI9341 + ESP32 WROOM

datodato788, 2025-09-27:
How do I calibrate the screen so that it moves properly?

datodato788, 2025-09-27:
https://drive.google.com/file/d/1CfkobBU-C2dcuZNGebrmtEGKzJZZ0Btb/view?usp=sharing

goku0x01, 2025-10-01:
> How do I calibrate the screen so that it moves properly?

just use a stylus pen with the tft display.

tester790, 2025-10-23:
anyone has any idea why is this happening and why is it in this orientation i was using the old_hardware.bin is there any solution 

[image]

goku0x01, 2025-11-06:
> anyone has any idea why is this happening and why is it in this orientation i was using the old_hardware.bin is there any solution
> 
> [image]

which esp32 are you using? the _old.bin is specifically for old esp32s